bluerobotics/BlueOS #102: ArduPilotManager fails when running on Desktop

rafaellehmkuhl, 2021-03-02:
```
File "companion-docker/core/services/ardupilot_manager/flight_controller_detector/Detector.py", line 5, in <module>
    from smbus2 import SMBus
ModuleNotFoundError: No module named 'smbus2'
```

It would be good to at least catch this.

patrickelectric, 2021-03-02:
Did you used a venv or installed the dependencies via setup.py ?
https://github.com/bluerobotics/companion-docker/blob/master/core/services/ardupilot_manager/setup.py

rafaellehmkuhl, 2021-03-03:
Setup.py directly

patrickelectric, 2021-03-03:
And it did not install smbus2 ? I have done a quick test here and it installed it

rafaellehmkuhl, 2021-03-03:
Hey, I probably missed something during the setup.py install. I runned it again and it indeed installed now.

closed 2021-03-03 by patrickelectric