bluerobotics/BlueOS #106: Install script fails at enabling Docker Daemon

rafaellehmkuhl, 2021-03-05:
This was noticed on a fresh Raspbian Lite installation, running on a RaspberryPi 4.

patrickelectric, 2021-03-05:
Can you provide the output of the installation script ?

patrickelectric, 2021-03-11:
@rafaellehmkuhl any update ?

rafaellehmkuhl, 2021-03-15:
Entire output of running /install/install.sh:

```bash
pi@raspberrypi:~ $ sudo curl -fsSL https://raw.githubusercontent.com/bluerobotics/companion-docker/master/install/install.sh | sudo bash
Starting hardware configuration.
Configuring BCM27XX board (Raspberry Pi 4)..
- Enable I2C, SPI and UART.
dtparam=i2c_arm=on
dtparam=spi=on
dtoverlay=spi1-3cs
dtoverlay=i2c4,pins_6_7
dtoverlay=uart1
- Set up kernel modules.
bcm2835-v4l2
i2c-bcm2835
i2c-dev
- Configure serial.
Checking for blocked wifi and bluetooth.
Checking for available space.
Checking for docker.
bash: line 38: docker: command not found
# Executing docker install script, commit: 3d8fe77c2c46c5b7571f94b42793905e5b3e42e4
+ sh -c apt-get update -qq >/dev/null
+ sh -c DEBIAN_FRONTEND=noninteractive apt-get install -y -qq apt-transport-https ca-certificates curl >/dev/null
+ sh -c curl -fsSL "https://download.docker.com/linux/raspbian/gpg" | apt-key add -qq - >/dev/null
Warning: apt-key output should not be parsed (stdout is not a terminal)
+ sh -c echo "deb [arch=armhf] https://download.docker.com/linux/raspbian buster stable" > /etc/apt/sources.list.d/docker.list
+ sh -c apt-get update -qq >/dev/null
+ [ -n  ]
+ sh -c apt-get install -y -qq --no-install-recommends docker-ce >/dev/null
E: Sub-process /usr/bin/dpkg returned an error code (1)
Cannot connect to the Docker daemon at unix:///var/run/docker.sock. Is the docker daemon running?
Going to install companion-docker version master.
Downloading and installing udev rules.
Downloading bootstrap
Cannot connect to the Docker daemon at unix:///var/run/docker.sock. Is the docker daemon running?
pi@raspberrypi:~ $ 
```

patrickelectric, 2021-03-15:
What is the operating system and version that you are running ? 
I'm doing my test with the latest raspbian: 2021-01-11-raspios-buster-armhf-lite
Have you done something before calling the install script ?
How are you connected to the internet ?
Please share any relevant information

rafaellehmkuhl, 2021-03-16:
Did update and upgrade apt repos before running the script. Will try again without it.

patrickelectric, 2021-03-19:
Unable to replicate it
[image]
[image]

rafaellehmkuhl, 2021-03-19:
Just tested here and it's working properly.
Problem was probably my `apt upgrade` before running the install script.

```bash
pi@raspberrypi:~ $ sudo su -c 'curl -fsSL https://raw.githubusercontent.com/bluerobotics/companion-docker/master/install/install.sh | bash'
Checking if network and remote are available.
Starting hardware configuration.
Configuring BCM28XX board (Raspberry Pi zero, 1, 2, 3)..
- Enable I2C, SPI and UART.
dtparam=i2c_arm=on
dtparam=spi=on
dtoverlay=spi1-3cs
dtoverlay=uart1
- Set up kernel modules.
bcm2835-v4l2
i2c-bcm2835
i2c-dev
- Configure serial.
Checking for blocked wifi and bluetooth.
Checking for available space.
Checking for docker.
bash: line 44: docker: command not found
# Executing docker install script, commit: 3d8fe77c2c46c5b7571f94b42793905e5b3e42e4
+ sh -c apt-get update -qq >/dev/null
+ sh -c DEBIAN_FRONTEND=noninteractive apt-get install -y -qq apt-transport-https ca-certificates curl >/dev/null
+ sh -c curl -fsSL "https://download.docker.com/linux/raspbian/gpg" | apt-key add -qq - >/dev/null
Warning: apt-key output should not be parsed (stdout is not a terminal)
+ sh -c echo "deb [arch=armhf] https://download.docker.com/linux/raspbian buster stable" > /etc/apt/sources.list.d/docker.list
+ sh -c apt-get update -qq >/dev/null
+ [ -n  ]
+ sh -c apt-get install -y -qq --no-install-recommends docker-ce >/dev/null
+ sh -c docker version
Client: Docker Engine - Community
 Version:           20.10.5
 API version:       1.41
 Go version:        go1.13.15
 Git commit:        55c4c88
 Built:             Tue Mar  2 20:18:46 2021
 OS/Arch:           linux/arm
 Context:           default
 Experimental:      true

Server: Docker Engine - Community
 Engine:
  Version:          20.10.5
  API version:      1.41 (minimum version 1.12)
  Go version:       go1.13.15
  Git commit:       363e9a8
  Built:            Tue Mar  2 20:16:18 2021
  OS/Arch:          linux/arm
  Experimental:     false
 containerd:
  Version:          1.4.4
  GitCommit:        05f951a3781f4f2c1911b05e61c160e9c30eaa8e
 runc:
  Version:          1.0.0-rc93
  GitCommit:        12644e614e25b05da6fd08a38ffa0cfe1903fdec
 docker-init:
  Version:          0.19.0
  GitCommit:        de40ad0
If you would like to use Docker as a non-root user, you should now consider
adding your user to the "docker" group with something like:

  sudo usermod -aG docker your-user

Remember that you will have to log out and back in for this to take effect!

WARNING: Adding a user to the "docker" group will grant the ability to run
         containers which can be used to obtain root privileges on the
         docker host.
         Refer to https://docs.docker.com/engine/security/security/#docker-daemon-attack-surface
         for more information.
Synchronizing state of docker.service with SysV service script with /lib/systemd/systemd-sysv-install.
Executing: /lib/systemd/systemd-sysv-install enable docker
Going to install companion-docker version master.
Downloading and installing udev rules.
Downloading bootstrap
master: Pulling from bluerobotics/companion-bootstrap
e12d26f75fe8: Pull complete 
1769c2da4b04: Pull complete 
9d2f919117d9: Pull complete 
7c18f801328d: Pull complete 
50a92cd408a6: Pull complete 
Digest: sha256:6ada4e438e16c999a9a15bf2910d36847efde58455fe0dd4876ddbeb30ede011
Status: Downloaded newer image for bluerobotics/companion-bootstrap:master
docker.io/bluerobotics/companion-bootstrap:master
unable to read startup.json file ([Errno 2] No such file or directory: '/root/.config/companion/startup.json'), reverting to defaults...
Attempting to pull an updated image... This might take a while...
Done
Starting core
Core started
Done
Starting network configuration.
Configuring companion avahi service
Configure hostname to companion
Installation finished successfully.
System will reboot in 10 seconds.
Connection to 192.168.0.51 closed by remote host.
Connection to 192.168.0.51 closed.
```

closed 2021-03-19 by rafaellehmkuhl