bluerobotics/BlueOS #128: install steps are creating bad urls

jaxxzer, 2021-03-26:
When using custom remote + version:

in [step 1](https://github.com/bluerobotics/companion-docker/blob/master/install/install.sh#L6) install.sh, we set `$REMOTE` to `github.com/branch`.
in [step 2](https://github.com/bluerobotics/companion-docker/blob/master/install/boards/configure_board.sh#L7) configure board, we set `$REMOTE` to `github.com/branch/branch`, this leads to curl 404

jaxxzer, 2021-03-26:
I do not know why this does not affect the default without environment REMOTE and VERSION being set:
`sudo su -c 'curl -fsSL https://raw.githubusercontent.com/bluerobotics/companion-docker/master/install/install.sh | bash'`

patrickelectric, 2021-03-26:
There is a difference between environment variables and script variables, can you provide the necessary steps to replicate this issue ?

jaxxzer, 2021-03-26:
> There is a difference between environment variables and script variables

yes this problem is confusing to me, I'm not sure the source. check [branch-install](https://github.com/jaxxzer/companion-docker/commits/branch-install) patch:

```
root@raspberrypi:/home/pi# export VERSION=branch-install
root@raspberrypi:/home/pi# export REMOTE=https://raw.githubusercontent.com/jaxxzer/companion-docker
root@raspberrypi:/home/pi# curl -fsSL https://raw.githubusercontent.com/jaxxzer/companion-docker/branch-install/install/install.sh | bash
Checking if network and remote are available.
Starting hardware configuration.
Detecting board type
Detected BCM27XX via device tree
***https://raw.githubusercontent.com/jaxxzer/companion-docker/branch-install/branch-install/install/boards/bcm_27xx.sh
***curl: (22) The requested URL returned error: 404
Checking for blocked wifi and bluetooth.
Checking for available space.
Checking for docker.
```

jaxxzer, 2021-03-26:
It looks like because `configure_board` is launched from inside `install`, the changes to the `REMOTE` variable in `install` are carried on to `configure_board`:

```
root@raspberrypi:/home/pi# export VERSION=branch-install-2
root@raspberrypi:/home/pi# export REMOTE=https://raw.githubusercontent.com/jaxxzer/companion-docker
root@raspberrypi:/home/pi# curl -fsSL https://raw.githubusercontent.com/jaxxzer/companion-docker/branch-install-2/install/install.sh | bash
Checking if network and remote are available.
Starting hardware configuration.
one ***next line we see $REMOTE is REMOTE+VERSION***
https://raw.githubusercontent.com/jaxxzer/companion-docker/branch-install-2
branch-install-2

Detecting board type
Detected BCM27XX via device tree
```

patrickelectric, 2021-03-26:
Check #131, the problem is the missing REMOTE and VERSION variable inside bcm_27xx

PR #129 (2021-03-26): Fixup
fix #128 
fix a problem with the spi0-dts download
closed 2021-03-26 by jaxxzer