bluerobotics/BlueOS #144: udpout autopilot connections are not working

jaxxzer, 2021-04-02:
The configuration is loaded, but I am not getting any data on the other side

```
root@companion:/# /home/pi/services/ardupilot_manager/main.py
Loaded settings from /root/.config/ardupilot-manager/settings.json:
{'endpoints': [{'argument': 14550,
                'connection_type': 'udpout',
                'place': '192.168.2.1'}]}
```

jaxxzer, 2021-04-02:
The default tcp 5760 port is working, the default and settings configured udp ports are not showing as in use:
```
root@companion:/home/pi# netstat -tulpn
Active Internet connections (only servers)
Proto Recv-Q Send-Q Local Address           Foreign Address         State       PID/Program name    
tcp        0      0 0.0.0.0:8080            0.0.0.0:*               LISTEN      972/python3         
tcp        0      0 0.0.0.0:22              0.0.0.0:*               LISTEN      555/sshd            
tcp        0      0 0.0.0.0:8000            0.0.0.0:*               LISTEN      959/python3         
tcp        0      0 0.0.0.0:5760            0.0.0.0:*               LISTEN      1158/ardusub        
tcp        0      0 0.0.0.0:9090            0.0.0.0:*               LISTEN      948/python3         
tcp        0      0 0.0.0.0:6020            0.0.0.0:*               LISTEN      1310/mavlink-camera 
tcp        0      0 0.0.0.0:9000            0.0.0.0:*               LISTEN      953/python3         
tcp6       0      0 :::22                   :::*                    LISTEN      555/sshd            
udp        0      0 0.0.0.0:68              0.0.0.0:*                           545/dhcpcd          
udp        0      0 0.0.0.0:50797           0.0.0.0:*                           1310/mavlink-camera 
udp        0      0 0.0.0.0:57554           0.0.0.0:*                           1310/mavlink-camera 
udp        0      0 0.0.0.0:5353            0.0.0.0:*                           397/avahi-daemon: r 
udp        0      0 0.0.0.0:60960           0.0.0.0:*                           397/avahi-daemon: r 
udp6       0      0 :::5353                 :::*                                397/avahi-daemon: r 
udp6       0      0 :::57648                :::*                                1310/mavlink-camera 
udp6       0      0 :::57371                :::*                                397/avahi-daemon: r 
root@companion:/home/pi# netstat -tupn
Active Internet connections (w/o servers)
Proto Recv-Q Send-Q Local Address           Foreign Address         State       PID/Program name    
tcp        0      0 192.168.2.2:22          192.168.2.1:45696       ESTABLISHED 1202/sshd: pi [priv 
udp        0      0 169.254.213.35:44515    8.8.8.8:53              ESTABLISHED 355/systemd-timesyn
```

rafaellehmkuhl, 2021-04-02:
The endpoints are added to mavlink-manager (e.g. mavproxy). One can se that by fetching them through get_available_endpoints, on the REST API. 

The problem is indeed the missing restart after adding new endpoints.

closed 2021-04-07 by rafaellehmkuhl