bluerobotics/BlueOS #146: Cable-guy is re-setting the ethernet connection while other services start

Williangalvani, 2021-04-02:
This is affecting the mavlink communication.
Once the connection got stabilished, lost, and regained.

Another time the udp communication was stabilished while ethernet was down, and bound to the docker IP:

```pi@companion:~ $ netstat -utpn
(Not all processes could be identified, non-owned process info
 will not be shown, you would have to be root to see it all.)
Active Internet connections (w/o servers)
Proto Recv-Q Send-Q Local Address           Foreign Address         State       PID/Program name    
tcp        0      0 192.168.2.2:22          192.168.2.1:46538       ESTABLISHED -                   
udp        0      0 169.254.199.46:43686    192.168.2.1:14550       ESTABLISHED -                   
udp        0      0 169.254.213.35:43191    8.8.8.8:53              ESTABLISHED - 
```

rafaellehmkuhl, 2021-11-23:
@Williangalvani to better understand this one:

It's just a matter of not changing the IP if it's the same as the currently set or we also need to deal with some race condition, like making sure cable-guy completes it's IP setting before starting the other services?

Williangalvani, 2021-12-13:
well, that is pretty good question... I think we should go with option # 1 at first so we don't slow the boot process any longer... If that isn't enough, we set up some kind of boot order or dependency system.

Williangalvani, 2025-01-10:
I haven't seen this in a very long time, so I'll close it for now.

closed 2025-01-10 by Williangalvani