bluerobotics/BlueOS #150: FirmwareDownload is failing minimum test requirement

patrickelectric, 2021-04-07:
@rafaellehmkuhl check https://github.com/bluerobotics/companion-docker/actions/runs/726942311

rafaellehmkuhl, 2021-04-07:
Apparently this is due to problems on the ardupilot firmware servers.
I had this locally already. From time to time it get's down or intermitent.

patrickelectric, 2021-04-07:
Should we handle it somehow ?

rafaellehmkuhl, 2021-04-08:
We can move to `urlopen` instead of `urlretrieve` and set a timeout. It's not guaranteed, but it helps with intermittent servers.

A more correct approach would also include separating unit tests from integration tests (this case), so we can easily identify when some test is not passing because of changes/downtime on external resources, instead of local changes.

rafaellehmkuhl, 2021-04-08:
Also, `urlretrieve` is going to be deprecated in the future, so moving it is would be good.
As we are dealing with small files in this case, there's no problem in moving without more modifications.

closed 2021-04-08 by patrickelectric