bluerobotics/BlueOS #154: Mavlink logging system

rafaellehmkuhl, 2021-04-08:
Currently we do have logging enable by default on MavlinkRouter (`--log <dir>`) and MAVProxy (`--state-basedir=<dir>`), although I think it's not correctly implemented on MAVProxy (should be `--logfile=LOGFILE`).

We also have the possibility for File endpoints on the Endpoint implementation, although neither MavlinkRouter nor MAVProxy allow them.

What do we want in the end?

I was thinking about the following possibility:
- Keep the default logging at /tmp
- Allow for File endpoints on all interfaces, and stream/log the data on the desired file. This would be a feature implemented on top of the two router interfaces, effectively adding functionality to them.

What do you guys think? @jaxxzer @Williangalvani @patrickelectric

closed 2021-06-04 by rafaellehmkuhl

patrickelectric, 2021-06-04:
@rafaellehmkuhl can you point the PR that fixes the issue ?

rafaellehmkuhl, 2021-06-04:
This issue was intended to track the File endpoint behaviour, but this was removed on #152