bluerobotics/BlueOS #171: ArduPilotManager not starting on Navigator

rafaellehmkuhl, 2021-04-20:
This is returned from the container:

```bash
root@companion:/# /home/pi/services/ardupilot_manager/main.py
Loaded settings from /root/.config/ardupilot-manager/settings.json:
{}
Traceback (most recent call last):
  File "/home/pi/services/ardupilot_manager/main.py", line 49, in <module>
    autopilot.run()
  File "/home/pi/services/ardupilot_manager/ArduPilotManager.py", line 39, in run                                                                            
    while not self.start_board(BoardDetector.detect()):
  File "/home/pi/services/ardupilot_manager/ArduPilotManager.py", line 103, in start_board                                                                   
    self.start_navigator()
  File "/home/pi/services/ardupilot_manager/ArduPilotManager.py", line 55, in start_navigator                                                                
    os.rename(temporary_file, firmware)
OSError: [Errno 18] Invalid cross-device link: '/tmp/czjfwdyqpktusebv-ardusub' -> '/root/.config/ardupilot-manager/firmware/ardusub'
```

rafaellehmkuhl, 2021-04-20:
@jaxxzer it's the first time I use APManager with the Navigator. Have you seen this before?

patrickelectric, 2021-04-20:
It appears that is something related to how os.rename works, doing a simlink or hardlink between two different devices, I would recommend to use move.

jaxxzer, 2021-04-20:
https://github.com/bluerobotics/companion-docker/pull/136 will fix it

jaxxzer, 2021-04-20:
Duplicate #135

closed 2021-04-20 by jaxxzer