bluerobotics/BlueOS #196: Do we need CORS?

rafaellehmkuhl, 2021-04-28:
Is there some use case were CORS is needed? 
Accessing a service on the Companion through another machine works without it?

Williangalvani, 2021-04-28:
I think the issue right now is that as we don't have something like nginx serving everythhing under the same port. And different ports could be considered different domains, causing CORS issues, but this needs to be checked.

rafaellehmkuhl, 2021-06-17:
We need CORS to fetch data from one service to another (e.g.: the main frontend fetching data from the services).

patrickelectric, 2021-06-17:
I believe that is time to have a web server for companion so, @rafaellehmkuhl please push the frontend template to a repository when possible

rafaellehmkuhl, 2021-06-17:
Will. I'm currently solving some typescript warnings, but will upload an image so you guys can take a feel.

rafaellehmkuhl, 2021-06-17:
Regarding have some service like nginx putting everybody under the same hostname, this won't solve for the need of CORS, as they will continue to be hosted on different ports (thus being considered different origins).

patrickelectric, 2021-06-17:
nginx Will redirect all cors related communication to a path

rafaellehmkuhl, 2021-06-17:
@Williangalvani remembered that we can put all services under the same URL, just different paths, so this should solve the problem of local services.

patrickelectric, 2021-06-17:
They can if they want, I don't see a problem with that

Williangalvani, 2021-06-17:
Cors can be enable on nginx itself. There's no need to implement it for each individual server

rafaellehmkuhl, 2021-06-17:
> Cors can be enable on nginx itself. There's no need to implement it for each individual server

Yes. Best solution.

rafaellehmkuhl, 2021-06-25:
Fixed by #351

closed 2021-06-25 by rafaellehmkuhl