bluerobotics/BlueOS #205: Cable-guy swagger is missing documentation for endpoints

patrickelectric, 2021-04-30:
Found in: d1183dc
[image]

It's missing /ethernet

rafaellehmkuhl, 2021-04-30:
It's not. Check under /v1.0/docs.

/docs alone shows only non-API endpoints (index, for example) and the available docs.

patrickelectric, 2021-04-30:
`/docs` should not exist since is not useful, this is also what the helper is pointing to. We should remove `/docs` or redirect it to `/docs/latest/`

rafaellehmkuhl, 2021-04-30:
It's useful since it points out the available versions of the API, and also the available non-versioned endpoints (like index).

On the other hand, helper should point to the latest API version, I agree.

rafaellehmkuhl, 2021-04-30:
[This issue](#204 ) is is aimed to solve that.

closed 2021-04-30 by patrickelectric

patrickelectric, 2021-04-30:
@rafaellehmkuhl can you fix your hyperlink ?

rafaellehmkuhl, 2021-04-30:
Fixed