bluerobotics/BlueOS #207: Core: Add general web application

patrickelectric, 2021-04-30:


rafaellehmkuhl, 2021-04-30:
You mean the frontend you're working on or another thing?

patrickelectric, 2021-04-30:
> frontend you're working

yes

Williangalvani, 2021-06-01:
This is interesting:
https://innovation.enova.com/building-an-embeddable-micro-frontend-with-vue-js/
(I'll edit this post with new stuff I find)

patrickelectric, 2021-06-03:
From a quick discussion between our technological investigation, sadly we found not valid alternative to create what we want to have in our final version with microservice/plugin-like system.

We agreed with the following solution:
- Create a single project for the frontend
- The project will be a monolithic project
    - This will result for now in a bit of duplicated code, since a good part of the actual development pages will be common code.
    - But this allow us to to move to a webcomponent solution or a runtime vue component compiler without much pain, since the first has a big development and maintenance cost and the second is not technological possible with typescript for now. But as said, both can be accomplished since a good part of the code will exist in the monolithic frontend. 
- We are going to use bootstrap
    - It's a popular and common UI framework and we have in house experience with @Williangalvani and @rafaellehmkuhl 
- As a consequence of using bootstrap we'll need to use vue2
    - Vue3 is not fully supported under bootstrap, for more information: https://github.com/bootstrap-vue/bootstrap-vue/issues/5196
    - Vue2 should not be a problem since Vue3 update should be a no pain process, @rafaellehmkuhl said that he can handle it without any trouble for us.

PR #342 (2021-06-22): Add main frontend
Main frontend for companion-docker

Setup is:
- Vue 2
- SASS (with dart-sass)
- Typescript
- PWA support
- Router (with history mode)
- Vuex
- Vuetify

Adds test job for installing and building the frontend project.

This fixes #207, although there are import considerations there we should open new issue for in the future (regarding dynamic plugins).

Image available for test under `rafaellehmkuhl/companion-core:frontend`
closed 2021-06-22 by Williangalvani