bluerobotics/BlueOS #238: Allow connection to hidden wifi network

rafaellehmkuhl, 2021-05-17:
Maybe a button "Connect to hidden network".

rafaellehmkuhl, 2021-05-18:
@patrickelectric do you know if a "hidden network" does not show up on the scan, or if it shows up but with a null SSID?

patrickelectric, 2021-05-18:
I am almost sure that it appears with an empty SSID

rafaellehmkuhl, 2021-05-19:
If that's the case, cool, this is covered by #244.
My neighborhood has one of these (network with null SSID) and it's  showing up and converting the SSID to [HIDDEN_SSID].

patrickelectric, 2021-05-19:
Ok, but is it possible to connect with it ? That's what the issue is about.

rafaellehmkuhl, 2021-05-19:
> Ok, but is it possible to connect with it ? That's what the issue is about.

Don't know because I don't have the password :rofl: 

Will open one here and test.

rafaellehmkuhl, 2021-05-19:
Just tested here and indeed if some network has "Wireless SSID Broadcast" disabled, the network appears in the scan, but the SSID comes as `null`.

Yet to test connection.

PR #383 (2021-07-21): Refactor wifi service
Definitely a big refactor, but a much needed one.
Among the changes:
- Introduces timeouts for WPA operations (they could be slow or busy sometimes, making requests fail)
- Everything is async now (YES!) (scan operations, which can take from 5 to 15 seconds do not block other requests anymore)
- Scan logic is refactored so in-progress scan results are used on new fetchs (e.g: 10 requests will not necessarily create 10 scan operations. If there's one scan happening when you make a request, its result will be used)
- Network-registering is now separated from network-connect [... cut]
closed 2021-07-21 by Williangalvani