bluerobotics/BlueOS #240: Default endpoint for Ground Station communication is not supported by MavlinkRouter

rafaellehmkuhl, 2021-05-18:
The "GCS Link" endpoint is of the type "udpin", which is not supported by MavlinkRouter.

Easy solution is to use a type that is supported by all MavProxy, MavlinkRouter and QGroundControl.

Williangalvani, 2021-05-18:
[image]

This works well for the rov while we don't have a udpbcast option

rafaellehmkuhl, 2021-05-18:
> [image]
> 
> This works well for the rov while we don't have a udpbcast option

only problem there is that we need to know the IP of the GCS machine on runtime, right?

Williangalvani, 2021-05-18:
yes. but while we work on it, I see no issue in using 192.168.2.1. This is the IP where the current companion assumes the GCS is.

patrickelectric, 2021-05-18:
I agree with it, but I believe that is more an QGC issue than ours since it should find the ROV over the network, and not hardcoding stuff and having a GCS as a server.

rafaellehmkuhl, 2021-05-18:
Agree with both on that.

What do you think should be our approach today?

Have in mind MavlinkRouter is our default router on core today.

patrickelectric, 2021-05-18:
I would recommend to hardcode the IP as suggested.

PR #256 (2021-05-21): core: ardupilot-manager: Support UDP Server on MavlinkRouter and fix GCS Link default endpoint
Fix #240
Fix #254
Fix #255
closed 2021-05-21 by patrickelectric