bluerobotics/BlueOS #267: core: version chooser: sha information is not useful

patrickelectric, 2021-05-21:
The sha is not our git repository sha, our IMAGE ID, or dockerhub sha code.
Where they come from, who we should use, where are we going ?
[image]

Williangalvani, 2021-05-21:
That _is_ the image id
[image]

patrickelectric, 2021-05-21:
Ok, but my point is that is not clear which sha is this, and how useful is this sha.
How we can compare it with github or dockerhub to know that we are running the latest version ?

Williangalvani, 2021-05-21:
yeah, dockerhub is weird.
`docker image inspect 72814f1ef1cd`
actually shows a "RepoDigests" key, but the contents don't match dockerhub.
Maybe we're seeing 72814f1ef1cd? I can get back to this later.

patrickelectric, 2021-05-21:
Ok, I believe that we should work to improve this.
I vote to add the information of our github tag, hash, build time and etc in the docker metadata and use it with others important shas in the webpage in an explicit and clear way.

Williangalvani, 2021-05-21:
Yeah, it seems to be somewhat doable with [labels](https://nickjanetakis.com/blog/docker-tip-25-adding-metadata-to-your-docker-images-with-labels), I'll do some experimentation with it.

Williangalvani, 2022-02-22:
fixed in #782

only displayed in pirate mode, only loaded for tags available both locally and remotely

closed 2022-02-22 by Williangalvani