bluerobotics/BlueOS #277: Pip is randomly and silently failing

Williangalvani, 2021-05-25:
This leaves us with a successfully-built image with failing services due to missing dependencies.

Probably related issues during build:
[image]
[image]

patrickelectric, 2021-05-26:
@Williangalvani just to make sure #277 does not fix that ?

Williangalvani, 2021-05-26:
> @Williangalvani just to make sure #277 does not fix that ?

well. I'm not 100% sure. But let's close this and we can re-open it if it shows up again.

closed 2021-05-26 by Williangalvani