bluerobotics/BlueOS #293: No dev tools available on the host os

rafaellehmkuhl, 2021-05-28:
I would suggest that somehow we provide minimum dev tools on the host os, so one can start developing with minimum effort.

First thing that come to my mind and is not available on RaspOS Lite is `git`.

@Williangalvani suggested that we have a separated script `install_dev_tools.sh` to keep that separated from the regular install procedures.

patrickelectric, 2021-06-01:
The problem with that is that we don't know the name of such tools under different OSs and also the package manager that is using, so such universal script is not possible. The only way is to provide a script for each OS/package manager.

rafaellehmkuhl, 2021-06-01:
Yeah...
Let's just write a wiki article indicating recommended tools on dev section. Seems more reasonable.