bluerobotics/BlueOS #302: ardupilot parameters are not saved after power cycle or companion reboot

jaxxzer, 2021-06-03:


Williangalvani, 2021-06-03:
This seems to be due to an issue we fixed in master a while ago:
https://github.com/ArduPilot/ardupilot/pull/15606

Williangalvani, 2021-06-03:
This is a build of master from right now (c61570697140633d0fb3d545ebeae87f4c5174d0)

[ardusub.zip](https://github.com/bluerobotics/companion-docker/files/6593596/ardusub.zip)

jaxxzer, 2021-06-03:
> is the build in "https://s3.amazonaws.com/downloads.bluerobotics.com/ardusub/navigator/ardusub" custom?

yes

jaxxzer, 2021-06-03:
> we should use ardusub master with navigator for now or backport this

ok, I can handle it.

Williangalvani, 2021-06-03:
well. master seems to be failing too, here.

Williangalvani, 2021-06-03:
[image]

patrickelectric, 2021-06-03:
The problem is the spaces around the arguments, I'll send PR. 
[image]

PR #305 (2021-06-03): core: ardupilot_manager: Fix spaces before argument
Fix #302
closed 2021-06-03 by patrickelectric