bluerobotics/BlueOS #307: Core: Add hotspot if it's not connected in a wifi network

patrickelectric, 2021-06-03:


rafaellehmkuhl, 2021-06-04:
Cool. I have the PR for this one almost ready, using hostapd. Can deliver next week.

patrickelectric, 2021-07-29:
Any update on that ?

rafaellehmkuhl, 2021-07-29:
The branch is working but we deprioritized as its not on the current roadmap. There are some wifi/hotspot enable-disable tests to do yet.

rafaellehmkuhl, 2021-12-13:
Reading a little more about IoT solutions, I came to the opinion that we should have an always-available wifi hotspot, instead of one that is available only when there no client-connection. This makes everything easier and more intuitive to the user (companion hotspot will always be there). For this to work, the raspberry wlan physical interface needs support for virtual-wlan, which it does.

Implementing this demands a little more work but will certainly pay off in terms of user experience.

More information (including implementation) can be found on the following links:

- https://github.com/lukicdarkoo/rpi-wifi
- https://thepi.io/how-to-use-your-raspberry-pi-as-a-wireless-access-point/
- https://blog.ithasu.org/2017/11/raspberry-pi-0w-as-both-wifi-client-and-access-point/
- https://blog.thewalr.us/2017/09/26/raspberry-pi-zero-w-simultaneous-ap-and-managed-mode-wifi/
- https://imti.co/iot-wifi/
- https://www.0xf8.org/2016/02/using-your-raspberry-pi-zeros-usb-wifi-adapter-as-both-wifi-client-and-access-point/

patrickelectric, 2022-07-13:
The wifi name should be unique per vehicle.

rafaellehmkuhl, 2022-07-13:
> The wifi name should be unique per vehicle. 

That would be interesting. Will add.

You mean being configurable by the user or by using a part-random name?

I'm a fan of both, but the first already solves the problem so will use that for now.

closed 2022-07-25 by patrickelectric