bluerobotics/BlueOS #313: InterceptHandler's emit method had typing problems

rafaellehmkuhl, 2021-06-08:
This method is present on both ArduPilotManager and CableGuy. [Discussion with loguru's author](https://github.com/Delgan/loguru/issues/449) (where this  method came from) led to a possible, although somewhat non-ideal solution.

rafaellehmkuhl, 2021-06-08:
Solution proposed by @patrickelectric:

```python
from logging import LogRecord
from types import FrameType
from typing import Any, List, Optional, Union

class InterceptHandler(logging.Handler):
    def emit(self, record: LogRecord) -> None:
        # Get corresponding Loguru level if it exists
        level: Union[int, str]
        try:
            level = logger.level(record.levelname).name
        except ValueError:
            level = record.levelno

        # Find caller from where originated the logged message
        frame, depth = logging.currentframe(), 2
        frame: Optional[FrameType]
        while frame and frame.f_code.co_filename == logging.__file__:
            frame = frame.f_back
            depth += 1

        logger.opt(depth=depth, exception=record.exc_info).log(level, record.getMessage())
```

patrickelectric, 2021-08-25:
@rafaellehmkuhl this was fixed, no ?

rafaellehmkuhl, 2021-08-25:
Yes.

closed 2021-08-25 by rafaellehmkuhl