bluerobotics/BlueOS #332: Minimize startup time

jaxxzer, 2021-06-11:
after applying power on latest master image, it took 120s for ping 192.168.2.2 to respond, and for ardupilot to connect, wheras it only takes 30s for the hostname service and docker to start with login prompt after boot output on a display.

Williangalvani, 2021-06-14:
for the record: It takes a little under 60s in my setup

patrickelectric, 2021-06-14:
It took 30 seconds to access the ssh here and more 15 seconds to have the mavlink connection.
Remember that the [SD card class](https://raspberrypi.stackexchange.com/questions/320/what-is-a-typical-boot-time-for-the-standard-debian-distribution-on-a-typical-sd) will interfere with the boot time.

I did a quick research and the following services are the most expensive in boot time:
```
12.620s dhcpcd.service
 7.340s docker.service
 7.052s hciuart.service
 2.192s dev-mmcblk0p2.device
```

All 4 are important for us so it's a bit trick to avoid it.

An investigation possibility is to replace the docker command with podman.

jaxxzer, 2021-06-18:
More detailed tests:

Rpi 3b+
companion 0.0.26
sandisk ultra sd card
first ping @192.168.2.2: 15s
ardupilot connection: 30s

companion-docker - raspbian lite image
```
pi@companion:~ $ sudo docker container ls
CONTAINER ID   IMAGE                                     COMMAND                  CREATED         STATUS         PORTS     NAMES
f59d5274a22b   bluerobotics/companion-core:master        "/bin/sh -c '/usr/bi…"   3 minutes ago   Up 3 minutes             companion-core
e5a4dbabead0   bluerobotics/companion-bootstrap:master   "/bin/sh -c /main.py"    13 days ago     Up 3 minutes             companion-bootstrap
```
lexar 1000x sd card
first ping @192.168.2.2: 60s
ardupilot connection: 60s

patrickelectric, 2021-11-19:
- This is a duplicated of https://github.com/bluerobotics/companion-docker/issues/185

closed 2021-11-19 by patrickelectric