bluerobotics/BlueOS #382: Deal with docker rate-limiting

Williangalvani, 2021-07-09:
Just got this:
docker.errors.APIError: 500 Server Error for http+docker://localhost/v1.41/images/create?tag=master&fromImage=bluerobotics%2Fcompanion-core: Internal Server Error ("toomanyrequests: You have reached your pull rate limit. You may increase the limit by authenticating and upgrading: https://www.docker.com/increase-rate-limit")

The rate limit right now is of 100 pulls in a 6h interval. There's more information at https://www.docker.com/increase-rate-limit.

patrickelectric, 2021-07-10:
Just to make sure from, from a single client or for a single docker image ?

patrickelectric, 2021-08-25:
Maybe we should keep track and warn our users ?
"You have requested to many requests to check for a new update, your actual limit is x of 100"

Williangalvani, 2021-09-24:
each manifest download counts as a pull
We do one manifest pull for each tag in a remote when checking available tags.

patrickelectric, 2021-11-10:
If'm not mistaken the plan was to have a "cache" for manifests or only fetch such manifest when necessary, no ?

patrickelectric, 2022-02-22:
@Williangalvani we have fixed that with latest rework right ?

patrickelectric, 2022-02-22:
#782

Williangalvani, 2022-02-22:
yes!

closed 2022-02-22 by Williangalvani