bluerobotics/BlueOS #390: Version Chooser frontend keeps loading when there's no internet connection on the network

rafaellehmkuhl, 2021-07-21:
As we move to the main frontend, the old one does not need to be fixed. Opening the issue so we track this change for the new one.

Williangalvani, 2021-07-22:
it does load here, but it takes 18s to do so. My guess is that the backend is taking a long time to time out.

rafaellehmkuhl, 2021-07-22:
From what I can remember I've waited more than that and it didn't stop, but will double check.

patrickelectric, 2021-08-25:
It would be nice to have a "no internet connection" message, a offline game would also be nice :t-rex: :eyes:

rafaellehmkuhl, 2021-08-25:
Just tested here 

> it does load here, but it takes 18s to do so. My guess is that the backend is taking a long time to time out.

Just tested here and after 1 minute the `companion-core` request hadn't completed yet.

rafaellehmkuhl, 2021-08-25:
I would add a suggestion of behavior here: if there's no internet connection, show the local images and the current image, and allow regular behavior with them.

rafaellehmkuhl, 2022-02-03:
Fixed with #770

rafaellehmkuhl, 2022-02-03:
Fixed on #770

closed 2022-02-03 by rafaellehmkuhl