bluerobotics/BlueOS #398: Ardupilot loop is slipping a lot, triggering internal errors.

Williangalvani, 2021-08-05:
This is likely due to the scheduler.
A couple of things helped so far:

 - i2c at 1MHz helps a lot, but not 100%, it is also potentially unsafe (we need to check all datasheets and pull-ups)
 - Custom 4.9 PREEMPT kernel: worked nicely with i2c bauds of 400khz, with minor slips
 
 things to test still:
   - use isolcpu to allow ardupilot to run on it's own core of the Pi.
   - disable interruptions at the core ardupilot is running?
   - nice -20
   - proper preempt-rt patch

I think we should try to steer away from using a custom kernel, but it is good to know it is a solution.

patrickelectric, 2021-11-10:
@Williangalvani that was fixed using a RT kernel right ?
Can you somehow document the process of creating a raspberry image with such kernel ?

patrickelectric, 2022-01-28:
@Williangalvani is this still valid ?

patrickelectric, 2022-03-17:
ping @Williangalvani

patrickelectric, 2022-03-25:
We are using a normal linux image with a lower loop frequency on ardupilot

closed 2022-03-25 by patrickelectric