bluerobotics/BlueOS #403: frontend: Add network webpage

patrickelectric, 2021-08-10:
- [ ] Network configuration
    - [ ] DHCP client
    - [ ] DHCP server
        - [ ] Show list of clients
    - [ ] Static IP 
- [ ] Speed test from browser 
- [ ] Speed test from the internet provider
- [ ] Wifi connection
- [ ] IP information about network interfaces
- [ ] Current speed for each network interface

rafaellehmkuhl, 2021-08-10:
For Beta I would remove the three speed tests. 
Although important for the final release, they don't seem to match the Beta goal ("perform basic functionality").

rafaellehmkuhl, 2021-08-30:
I split this issue into three other (#448, #443 and #442) dedicated to each matter (wifi, ethernet and speed).
Closing this one.

closed 2021-08-30 by rafaellehmkuhl