bluerobotics/BlueOS #413: Notification system is crazy when ROV turns off

patrickelectric, 2021-08-14:
[image]

rafaellehmkuhl, 2021-08-14:
Hfuehfuehduhe

Yeah

This is happening because different requests are intercalated.

Maybe we should do like you've said before and just merge all equal notifications.

patrickelectric, 2021-08-14:
We may want also to have a high level notification message saying that communication may be lost. 
Maybe something on the header ?

rafaellehmkuhl, 2021-08-14:
I was just thinking about that.

If there's no communication, all the other messages do not make sense, so we could add a COMM_LOST message, that, when present, suppress the push of any other messages.

patrickelectric, 2021-08-14:
That sounds like a good approach

rafaellehmkuhl, 2021-08-14:
> We may want also to have a high level notification message saying that communication may be lost.
> Maybe something on the header ?

Didn't get the header idea, though. What do you mean?

patrickelectric, 2021-08-14:
In main webpage we could have something that indicates that the connection with the ROV was lost besides a high priority notification on the notification trail, like:
[image]

rafaellehmkuhl, 2021-08-15:
Cool! Will do!

patrickelectric, 2021-11-19:
- Done in https://github.com/bluerobotics/companion-docker/pull/608

closed 2021-11-19 by patrickelectric