bluerobotics/BlueOS #417: core: ardupilot manager:  Error updating endpoints: 'MAVLinkRouter' object has no attribute '_master_endpoint'

patrickelectric, 2021-08-18:
| Date | What     |  Text                                                                                                                               |
|-----------------|-----------|--------------
021-08-18 21:49:43.028 | INFO     | ArduPilotManager:start_board:180 - Board in use: Serial.                                                                                       
2021-08-18 21:49:43.053 | INFO     | ArduPilotManager:add_new_endpoints:235 - Adding endpoint 'GCS Link' and saving it to the settings file.                                        
2021-08-18 21:49:43.056 | INFO     | settings:save:104 - Updating settings file: /root/.config/ardupilot-manager/settings.json                                                      
2021-08-18 21:49:43.059 | INFO     | mavlink_proxy.AbstractRouter:exit:86 - Tried to stop router, but it was already not running.                                                   
2021-08-18 21:49:43.060 | ERROR    | ArduPilotManager:_update_endpoints:222 - Error updating endpoints: 'MAVLinkRouter' object has no attribute '_master_endpoint'                  
2021-08-18 21:49:43.064 | INFO     | ArduPilotManager:add_new_endpoints:235 - Adding endpoint 'MAVLink2Rest' and saving it to the settings file.                                    
2021-08-18 21:49:43.066 | INFO     | settings:save:95 - No new data. Not updating settings file.                                                                                    
2021-08-18 21:49:43.067 | INFO     | mavlink_proxy.AbstractRouter:exit:86 - Tried to stop router, but it was already not running.                                                   
2021-08-18 21:49:43.068 | ERROR    | ArduPilotManager:_update_endpoints:222 - Error updating endpoints: 'MAVLinkRouter' object has no attribute '_master_endpoint'                  
Could not open conf file '/etc/mavlink-router/main.conf' (No such file or directory)                                                                                                
2021-08-18 21:49:43.118 | DEBUG    | asyncio.selector_events:__init__:59 - Using selector: EpollSelector                                                                            
Open UART [4] /dev/autopilot *                                                                                                                                                      
UART [4] speed = 115200                                                                                                                                                             
Open UDP [5] 127.0.0.1:14000                                                                                                                                                        
Open UDP [6] 192.168.2.1:14550                                                                                                                                                      
2021-08-18 21:49:43.297 | INFO     | uvicorn.server:serve:64 - Started server process [32]                                                                                          
2021-08-18 21:49:43.299 | INFO     | uvicorn.lifespan.on:startup:26 - Waiting for application startup.                                                                              
2021-08-18 21:49:43.305 | INFO     | uvicorn.lifespan.on:startup:38 - Application startup complete.

rafaellehmkuhl, 2021-08-18:
Yup.
The ardupilot-manager initialization is currently a mess. I need to give it some work.
Besides the errors, that are not affecting functionality but are making the log unreadable, I also need to capture the mavlink-router stdout/stderror with the logger, so we get it properly logged/formated.

rafaellehmkuhl, 2022-01-19:
Behaviour changed at some point in time:

[image]
[image]

(sorry for the prints, having trouble copying from tmux on mac)

closed 2022-01-19 by rafaellehmkuhl