bluerobotics/BlueOS #418: propose: Change in Companion's default communication behavior

rafaellehmkuhl, 2021-08-18:
We are currently using a hardcoded UDP client endpoint (`192.168.2.1`) for the top machine on companion-docker. This is the same behavior we have on companion0. There are some problems with this approach. Two are the main ones:
- We need the user to configure it's top machine with a static IP
- With `mavlink-router`, if there's no UDP server online with this hardcoded IP on the local network, it will fail to route all mavlink communication

My proposal for a new behavior is the following:
- Remove the hardcoded IP from companion
- Make companion start as a DHCP server by default if it hasn't received a DHCP handshake from an external route. This allows someone to just connect companion with an ethernet cable to it's top computer, which will receive a dynamic IP and enter companion's network
- Make companion start a Wifi hotspot with a DHCP server whenever there's no known network around for more than 10 seconds. This allows the same behavior as the ethernet DHCP server, but with wifi.

rafaellehmkuhl, 2021-08-19:
I really, really dislike needing to set up the connection manually in QGC.

_Originally posted by @Williangalvani in https://github.com/bluerobotics/companion-docker/issues/419#issuecomment-902089507_

Another possibility is to do the following:
1. Ask the user to add an UDP client endpoint on companion pointing to the top machine's IP
2.a. Add some logic on companion to check if the IP exists on the network, and if not, disable this endpoint
2.b. Work on `mavlink-router` to prevent that behavior

patrickelectric, 2021-08-19:
We should also think about the automatic connection / discovery of the companion computer via the GCS. 
As reference: https://discuss.px4.io/t/qgc-companion-computer-integration-idea/22831

Williangalvani, 2021-08-19:
We should take some time to think about this, considering multiple clients, boats, and remote clients as well...

So.. some loose requirements, it should:
 - [ ] be as plug and play as possible
 - [ ] support additional clients connecting with no further setup
 - [ ] support  multiple links/redundancy (not sure how this works) for boats.
 - [ ] support remote links for boat, preferably including both telemetry and video.
 
 
I would make the pi a dhcp by default, so there's no need to setup the fxti.

Then we should make it possible to find all qgc instances in the network, and just send the streams to these instances. Ideally, mDNS would work. other options are broadcasting like the Ping360, or even rudely scanning for open ports.

For remote access, I vote we add a server so we don't need to deal with NAT issues. 
I think redundancy is somewhat embeded in QGC, but it needs testing.

Williangalvani, 2021-08-19:
We agreed that the best options is to implement mdns discovery in QGC. It is possible to add additional metadata to the hosts, so we can advertise something like "companion.local" with additional data "mavlink_udp_server_at_port: 14450", for example.

patrickelectric, 2021-08-19:
Notes from the meeting on 19/08/2021 at 14:00 GMT-3.

We agreed that mdns could fix all our problems and remove code from any workaround in companion, making it easier to maintain and having a free functionality. 

Doing a small investigation we found that the avahi xml description file can have a `txt-record` that can be used to send data to mdns clients.

cat /etc/avahi/services/companion.service 
```xml
<?xml version="1.0" standalone='no'?>
<!DOCTYPE service-group SYSTEM "avahi-service.dtd">

<service-group>
  <name replace-wildcards="yes">%h</name>
  <service>
    <type>_http._tcp</type>
    <port>80</port>
    <txt-record>path=MinhaBatata</txt-record>
    <txt-record>path2=MinhaBatata2</txt-record>
  </service>
</service-group>
```

Browser:
[image]

There is a Qt library that has an avahi implementation and can be easily integrate in QGC.
https://github.com/nitroshare/qmdnsengine

patrickelectric, 2021-08-19:
As fields, I believe that the best option is to have two fields:
- one for the connection type: UDP Server/Client and TCP Server/Client
- and the second for port number: 1234, 14550 

It may be good to have an specific field for vehicle identification, to make sure that the avahi server is indeed a mavlink vehicle.

```xml
<txt-record>is_mavlink_vehicle=true</txt-record>
<txt-record>mavlink_connection_type=UDPServer</txt-record>
<txt-record>mavlink_connection_port=14550</txt-record>
```