bluerobotics/BlueOS #425: Is `callPeriodically` a concern?

rafaellehmkuhl, 2021-08-20:
I'm concerned about:
```
let potato = {
    a: async () => console.log("Potato")
}

callPeriodically(potato.a, 1000)

# somehow potato is dead
```

What happens if the object that callPeriodically is using dies or the variables used inside potato.a dies ?

_Originally posted by @patrickelectric in https://github.com/bluerobotics/companion-docker/pull/412#discussion_r692170200_

patrickelectric, 2022-03-04:
callPeriodically should receive a [WeakRef](https://developer.mozilla.org/en-US/docs/Web/JavaScript/Reference/Global_Objects/WeakRef) and check if it's still available, otherwise it should kill itself.

Williangalvani, 2023-01-23:
I started fixing this at https://github.com/bluerobotics/BlueOS-docker/pull/1316
There are a couple left, but most of them are always consumed by something in the ui

patrickelectric, 2024-06-19:
Fixed by #2430

closed 2024-06-19 by patrickelectric