bluerobotics/BlueOS #430: ardupilot-manager: No validation on ELF firmwares

rafaellehmkuhl, 2021-08-25:
We currently don't have validation of ELF firmwares on installation.

The old approach (checking the return code on the `--help` command) is not compatible with stable SITL firmwares, since they use to return 1 on `--help` (yep), and so it was removed.

The correct approach would be to use the Ardupilot's firmware validation tool, but it's currently broken for SITL firmwares. We should fix it and implement it on ardupilot-manager.

patrickelectric, 2021-08-25:
The issue of the problem was merged here: https://github.com/ArduPilot/ardupilot/pull/17573
I believe that only 4.1+ or 4.2+ have this fix.

rafaellehmkuhl, 2021-08-25:
Yep. Tried and DEV looks nice.

rafaellehmkuhl, 2021-10-10:
@patrickelectric we already implemented ELF validation in the sense that we check for both the system's architecture and the specified platform.

Do you see any other validation that we should add or can this issue be closed?

closed 2023-02-20 by patrickelectric