bluerobotics/BlueOS #448: frontend: Add network speed test

rafaellehmkuhl, 2021-08-30:
- [ ] Allow checking network speed
  - [ ] with current wifi connection
  - [ ] with current ethernet connection
  - [ ] with internet
  - [ ] with [iperf3](https://iperf.fr/) or similar tool

rafaellehmkuhl, 2022-03-14:
@patrickelectric should we close this one or just check all besides the iperf one?

patrickelectric, 2022-03-15:
That was not entirely by #843, we still need to add a network test to get the internet speed on the vehicle.

Williangalvani, 2022-04-26:
something like this would be nice later on
[image]

patrickelectric, 2022-08-19:
This appears to be a great library to use and have real-time updates on the speed test on the frontend: https://pypi.org/project/aiotestspeed/

patrickelectric, 2023-01-03:
Reference #1271

closed 2023-07-06 by patrickelectric