bluerobotics/BlueOS #452: ardupilot-manager: Pixhawk4 vs Pixhawk1

rafaellehmkuhl, 2021-08-31:
Apparently Pixhawk4 has a different `board_type` but uses the same `board_id` as Pixhawk1.

This is making the firmware-manager think they are the same board, but the Ardupilot Upload tool does not proceed with the firmware upload, thus reaching the timeout.

As we are probably going to have Pixhawk4 as a backup for Pixhawk1 (end of life), we should check for those differences and support it.

jaxxzer, 2021-08-31:
[image]

patrickelectric, 2021-08-31:
Just to make it clear, is this a problem related to our code or ArduPilot upload tool ?

rafaellehmkuhl, 2021-08-31:
Our code.
The upload tool is correctly recognizing that this is a different board and we should use a proper firmware.
We think it's the same board (for us, any serial board is the same, currently).

rafaellehmkuhl, 2021-08-31:
One thing we could do is to use the upload tool when detecting serial boards to correctly identify it's `board_id`, and set the correspondent board.

rafaellehmkuhl, 2021-08-31:
I've just checked and their `board_id` is different. Pix1 is 9 and Pix4 is 50. Our current approach on the firmware manager is good, we just need to fix the serial board detector to treat different boards different.

PR #866 (2022-03-16): core: ardupilot-manager: Support Pixhawk4 (+ other improvements)
- Fully support Pixhawk4 (add support for firmware management and correct board detection);
- Allow toggling between different connected boards (specially useful for development and testing);
- Disallow firmware management for unsupported board (not Pixhawk1 nor Pixhawk4), preventing bad firmware uploads;
- Improve board detection, preventing bootloader from being as a fully booted board;
- Prevent ardupilot binaries being left running by pruning them all (and not only the one running);
- Make mavlink-router behavior more predict [... cut]
closed 2022-03-16 by patrickelectric