bluerobotics/BlueOS #457: logs: Save Pixhawk/Serial logs on Companion

rafaellehmkuhl, 2021-09-01:
Right now Pixhawk logs are not saved on Companion, so users don't have access to it trough Log Browser.

patrickelectric, 2023-01-03:
@Williangalvani what is the status of this issue ? We did some telemetry changes on #1226 ?

ES-Alexander, 2023-01-13:
I tried setting `LOG_BACKEND_TYPE` to MAVLink and seemingly got some very inconsistent behaviour from the Log Browser:

https://user-images.githubusercontent.com/25898329/212253723-b2a1eb45-9493-4b94-b057-e118aad21ee7.mov

Regardless of how well the live-stream-to-autopilot approach works, I still think it's worth the mavftp approach being implemented too, since if someone uses lots of CPU or memory for other things then maybe they don’t want to stream logs live, but they may want to be able to fetch them off the autopilot without needing to open their enclosure and access the logs via QGC.

patrickelectric, 2024-06-19:
Already done via tlog, but will have full support to binlogs with mavftp mavftp #926

closed 2024-06-19 by patrickelectric