bluerobotics/BlueOS #464: core: frontend: Show only firmwares that work

rafaellehmkuhl, 2021-09-03:


Williangalvani, 2021-09-03:
This is actually a non-issue. The Navigator will only be released once we have a stable firmware for it.

rafaellehmkuhl, 2021-09-03:
Humnnnn, ok.
So this is what I was not understanding @jaxxzer ?? :rofl: 

And when this happen will we remove the 'STABLE-4.0.3' sub release that is currently on the manifest?

patrickelectric, 2021-09-04:
We can just fix that after we release navigator officially, from now on its
not necessary since we don't know which ArduPilot version will be
compatible with or final navigator revision. We could remove invalid
versions for the current navigator revision, but as I said, doesn't sound
necessary.

rafaellehmkuhl, 2021-09-06:
Cool. Should we leave this issue open to remind us or do we close it for now?

patrickelectric, 2021-09-06:
I would recommend to set it as a final release milestone

patrickelectric, 2023-01-03:
Was that fixed ?

rafaellehmkuhl, 2023-01-03:
> Was that fixed ?

I think @Williangalvani is better informed around the Navigator firmware situation to comment on that.

ES-Alexander, 2023-10-26:
I think #1152 is a better coverage of this issue, so I think this one should be closed as a duplicate.
The problem is not specific to Navigator.