bluerobotics/BlueOS #642: core: service: add simulated camera based on SITL

ES-Alexander, 2021-11-05:
Could be kept super simple and low res, but would be really cool and useful to have a camera stream to go along with the onboard SITL option.

Would mean new users could do some basic piloting practice before using a real vehicle, and would also make it easier to do testing. Would be extra awesome if there was some way of specifying a simulated current to fight against, but I expect that's best handled in a web interface (either as settings for the SITL flight controller, or as an overlay to a web GCS).

The whole simulated camera would also likely be easier in a web GCS, because it could send geometry instead of pixels to the display, which are then rendered by the browser...

Williangalvani, 2021-11-05:
This could help:
`gst-launch-1.0 ximagesrc use-damage=0 ! video/x-raw,framerate=15/1 ! videoscale method=0 ! video/x-raw,width=800,height=600  ! queue ! videoconvert ! x264enc  tune=zerolatency bitrate=50000 ! video/x-h264, profile=baseline ! rtph264pay config-interval=1 name=pay0 pt=96 ! udpsink host=127.0.0.1 port=5600`

it captures the current x server and encodes it to h264, then sends to qgc in the same machine.

Williangalvani, 2021-11-05:
So we could have a slim version of [bluesim built for the pi4](https://github.com/godotengine/godot-proposals/issues/988), connect sitl to that, and stream its output to qgc

ES-Alexander, 2021-11-05:
Definitely some cool ideas :-)
Although I wonder if it would be better to have this kind of thing in an extension, since people who don't care about simulation might not want the bluesim build taking up space on their SD...

patrickelectric, 2021-11-05:
> So we could have a slim version of [bluesim built for the pi4](https://github.com/godotengine/godot-proposals/issues/988), connect sitl to that, and stream its output to qgc

YES PLEASE YES YES
[image]