bluerobotics/BlueOS #663: Docker doesn't shutdown properly when rebooting

Williangalvani, 2021-11-17:
This causes the PWM to act up when using a Navigator board, as ardusub is not stopped properly.

patrickelectric, 2022-08-05:
Is there any way for us to fix that ?

patrickelectric, 2023-01-03:
Is this still an issue ?

Williangalvani, 2023-01-03:
The PCA issue is fixed AFAIR.
we could still have issues with corrupted files, as I think the containers are being killed instead of gracefully shut down (which they just don't know how to do).

patrickelectric, 2023-01-03:
Can you put more in details how it was fixes ? Or just not replicable ?

Williangalvani, 2023-01-03:
I haven't seen this anymore. This is the relevant PR: https://github.com/ArduPilot/ardupilot/pull/18635

It could also be that I just haven't rebooted the vehicle with motors on anymore (which is silly and shouldn't be allowed anyway).

closed 2023-01-03 by patrickelectric