bluerobotics/BlueOS #668: frontend: wifi: display status of the network connection procedure

patrickelectric, 2021-11-18:
We could inform what is the status of connection while doing it
[image]

rafaellehmkuhl, 2021-11-18:
With the `wpa_supplicant`, as it is today, this would demand a really hard work, since we are not using the events interface, but only fetching on demand.

I was taking a look at alternatives some days ago and found [this library ](https://github.com/seveas/python-networkmanager)which could be a great substitute for what we currently have and also facilitate the introduction of such features.

khancyr, 2022-02-24:
why not get rid of wpa_supplicant and use network-manager with the XML-RPC call ? it is really performant and simple to do in python even without python-networkmanager !

rafaellehmkuhl, 2022-02-24:
> 

Hey khancyr! What's up?

I'm also in favor of getting rid of the wpa_supplicant layer! It's a rework we should do at some point.

I was taking a look at XML-RPC calls. Really cool! Didn't know about them. Does network-manager have an XML-RPC server?

Another good alternative is [the `iwutil` module on `pyroute2`](https://github.com/svinota/pyroute2/blob/master/pyroute2.core/pr2modules/iwutil.py). While refactoring part of our ethernet manager (which uses `pyroute2`) I noticed their IW module is already very mature.

rafaellehmkuhl, 2022-02-24:
> why not get rid of wpa_supplicant and use network-manager with the XML-RPC call ? it is really performant and simple to do in python even without python-networkmanager !

> 

Hey khancyr! What's up?

I'm also in favor of getting rid of the wpa_supplicant layer! It's a rework we should do at some point.

I was taking a look at XML-RPC calls. Really cool! Didn't know about them. Does network-manager have an XML-RPC server?

Another good alternative is [the `iwutil` module on `pyroute2`](https://github.com/svinota/pyroute2/blob/master/pyroute2.core/pr2modules/iwutil.py). While refactoring part of our ethernet manager (which uses `pyroute2`) I noticed their IW module is already very mature.

patrickelectric, 2026-08-21:
Fixed by #676

closed 2026-08-21 by patrickelectric