bluerobotics/BlueOS #677: 5GHz wifi connections unstable

rafaellehmkuhl, 2021-11-20:
Apparently [there's some problem regarding Pi4 and 5GHz wifi networks](https://forums.raspberrypi.com/viewtopic.php?t=246170).
I've tried connecting to such networks here (from different routers) and I always have the same behavior: I can connect, but the connection is very unstable, and keeps dropping/reconnecting each 5-10 seconds.

According to the linked discussion, maybe changing the kernel version can help with that.

patrickelectric, 2021-11-22:
The 5G network does not appears on my rasp.

rafaellehmkuhl, 2021-11-22:
> The 5G network does not appears on my rasp.

Yeah, it's the same here.
They appear sometimes, but most of the time they are not there.

On the beggining I was suspicious there could be something with my patch, that's why I did the extended tests, but the same behavior can be seen on a brand new rasp with beta4 or older releases.

rafaellehmkuhl, 2021-11-22:
> Set the country code under: `/etc/wpa_supplicant/wpa_supplicant.conf`
> 
> ```
> ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev
> update_config=1
> country=BR
> ```
> 
> That works for me

After connecting to them, the connection is stable or keeps dropping?

rafaellehmkuhl, 2021-11-22:
From the weekly-meeting, @patrickelectric said he doesn't have unstability problems after changing country-code, so I assume that's it and #683 is enough.

closed 2021-11-22 by rafaellehmkuhl