bluerobotics/BlueOS #678: core: No internet connection on the docker container

rafaellehmkuhl, 2021-11-20:
In some cases, which I couldn't filter out yet, there's no internet connection on the `core` container.

I can ping internet websites on the host, but can't inside the container.

This leds to impossibility of fetching online images on version-chooser, among other things.

patrickelectric, 2021-11-22:
Do you have an image or a sdcard that replicates the problem ?
Something that we could use to replicate and investigate the problem

rafaellehmkuhl, 2021-11-22:
I will check the rasp with the problem to see if it persisted after the reboot, as we had a power down yesterday.

patrickelectric, 2021-11-22:
If possible, please provide a list of steps to replicate the issue.

rafaellehmkuhl, 2021-11-22:
The problem didn't persist after reboot. I will perform some tests to see if I can reproduce it again.

rafaellehmkuhl, 2021-11-24:
Reproduced the problem. It occurs when:

- Pi is connected via both wifi and ethernet
- Wifi has internet connection but ethernet doesn't
- Accessing companion using ethernet IP

This is probably our main use case, so it's important that it be solved.

maphstr, 2022-01-04:
Noticed this and thought i should mention it to you. This has been an issue for us on the old companion as well. It is solved by using 'metric' in /etc/dhcpcd.conf. 
`interface wlan0
metric 200

interface eth0
metric 300`

will prioritize wlan0 for internet. This is a permanent setting and if for some reason wlan0 doesn't have internet and eth0 does, you will have the same problem. But that might be a less frequent occurrence.