bluerobotics/BlueOS #680: Allow changing companion.local to something else

patrickelectric, 2021-11-22:
Users can have multiple vehicles on the same network, we should allow them to change the mdns name.

rafaellehmkuhl, 2021-11-22:
This use-case came earlier then expected haha

patrickelectric, 2021-11-22:
We should have a single docker to take care of mdns/avahi for us.

rafaellehmkuhl, 2021-12-02:
I would also add the possibility of giving a "pretty" name to each companion, and put that on some easy place (maybe together with the git info?). This way users can identify each ROV by how they call them already (Rufus? Steve? BlueWale?).

rafaellehmkuhl, 2021-12-02:
From an external discussion with @patrickelectric and @Williangalvani came the idea of making `companion.local` point to a "companion select page" instead of the regular frontend. This page would be the same for all companions and would list all companions available on the network, with their respective mdns as links to their proper frontends.

This way we avoid other problems like one companion fetching data from the other because of the common `companion.local` address.

Williangalvani, 2021-12-06:
from my tests, it looks like a restart of the avahi service is required in order to change the hostname. You can change services on-the-fly, but changing the hostname requires a restart. so we may really need that to be on it's own container, or somehow restart it in the host (if we just just bind the avahi config into core)

Williangalvani, 2021-12-06:
Changing `host-name=` in /etc/avahi/avahi-daemon.conf followed by a system restart seems to be enough for updating the published hostname.

patrickelectric, 2024-03-22:
This is now possible

closed 2024-03-22 by patrickelectric