bluerobotics/BlueOS #689: Current state of multiple-cameras support

Williangalvani, 2021-11-24:
On QGC: a patch like [this](https://github.com/mavlink/qgroundcontrol/compare/master...Williangalvani:joystickcamera?expand=1) (but hopefully better) is required to enable using the joystick to change cameras:
[image]

The camera streams need to be set to 
udp://192.168.2.1:XXXX, where XXXX is any free udp port (I recommend udp://192.168.2.1:5600 and udp://192.168.2.1:5605)

MavlinkCameraManager is having issues communicating with the current settings, we should consider temporarily changing it to `mavlink-camera-manager --default-settings BlueROVUDP --mavlink udpout:192.168.2.1:14550` so it tries to communicate to qgc directly.

Here I used the PS4 joystick and used the "click on touchpad" button to switch streams, as it is button 20 which we can't use for ardusub commands.

https://user-images.githubusercontent.com/4013804/143295102-615cdb05-dfc1-4bbb-a79f-f0f68fa0718a.mp4

Williangalvani, 2022-01-07:
https://github.com/mavlink/qgroundcontrol/pull/10106 exposes the camera controls to the joystick

Williangalvani, 2022-01-11:
this seems to affect QGC's hability to download vehicle parameters...

patrickelectric, 2022-08-02:
@Williangalvani what is the status of this issue ?

Williangalvani, 2022-08-02:
I believe all in here has been addressed. I'll close this.

closed 2022-08-02 by Williangalvani