bluerobotics/BlueOS #691: core: frontend: Getting backend offline messages while doing updates or anything a bit more heavy

patrickelectric, 2021-11-30:
[image]

patrickelectric, 2022-06-09:
Possible fixed by #1037

rafaellehmkuhl, 2022-06-09:
> Probably fixed by #1037

Because the services will be unavailable, and we will be on the Nginx `/status` case?

rafaellehmkuhl, 2022-06-09:
Humn, ok. You're probably right.
The problem it doesn't solve is the case where there's a version update happening, right?

patrickelectric, 2022-06-09:
Nops.

patrickelectric, 2022-08-05:
I believe that there is no way around if the system stop answering when dealing with high load.
We may want to increase the timeout or change the behavior when loosing timestamps..

rafaellehmkuhl, 2022-08-05:
We can have a store variable for the axios-backend instance that is updated by the version-chooser saying the system is under an update. This way, we can just block all the notifications and change everything for a "Updating system" notification.