bluerobotics/BlueOS #779: core: frontend: version chooser and firmware services surprisingly slow to load

ES-Alexander, 2022-02-05:
### Current behaviour

These pages don't seem to get very much information, but it seems to take a long time for them to load, which can be frustrating. The current spinning logo is definitely better than nothing.

### Expected or desired behaviour

Either somehow reduce the time between "user requests information" and "information appears" (i.e. fetch or process faster), or display the options one at a time as they get returned. The primary issue is the _feeling_ of "nothing is happening", although reducing unnecessary waiting time where possible is always a benefit.

### Prerequisites

- [X] I have checked to make sure that a similar request has not already been filed or fixed.

rafaellehmkuhl, 2022-02-07:
Both cases are probably because they depend on external information. Firmware-manager fetches ArduPilot's manifest, while version-chooser fetches dockerhub.

Are you experiencing slow load on the entire firmware-manager page, or just when you click on the selector to see the available firmwares?

For the case of version-chooser, it would benefit from loading both components separately and show the local images when they are already ready, instead of waiting for the dockerhub ones. @Williangalvani is currently working on a new version-chooser page and can discuss that better.

patrickelectric, 2022-02-07:
Can't we store the ArduPilot manifest file and have an update button on the UI for the user to use it when necessary ? 
Or update it on the background periodically or once the service start and has internet.

rafaellehmkuhl, 2022-02-07:
I'm in favor of having it fetched before the user needs it, but against having an update button (it's a file that changes once a week or less).

patrickelectric, 2022-02-07:
> I'm in favor of having it fetched before the user needs it, but against having an update button (it's a file that changes once a week or less).

The file is updated everyday IIRC, my point is that we should store the older file while a new one is not available (via head last-modified field), or no internet connection and etc. At least we are going to have a populated UI.

Williangalvani, 2022-02-07:
the issue in version-chooser is that fetching the sha takes one request per image. we don't actually need the sha of images not available locally (as there's nothing to compare against), so I'll stop fetching these.

rafaellehmkuhl, 2022-02-07:
From firmware management:

Problems:
- User is able to click the version menu even without version loaded
- It takes some (or a lot of) time to load the available firmwares

Solutions:
- Disable version select menu when fetching
- Fetch and store manifest file on ardupilot-manager install
- Use this file for next firmware-versions-fetching
- Allow user to update manifest file
- Auto-update manifest file with some frequency