bluerobotics/BlueOS #786: bug: sidebar scrollbar broken

ES-Alexander, 2022-02-10:
### Bug description

When the sidebar menu options are expanded (or if the window is vertically shrunk) they go off the bottom of the browser. A scrollbar only seems to appear when the options are taller than the screen or something, instead of taller than the current window view, and the scrollbar that appears still doesn't give access to all the options/doesn't show the buttons at the bottom.

[image] [image]

### Steps to reproduce

Method 1:
1. Expand both options in the sidebar while on a kinda short screen
2. Fail to access all the tools / the buttons at the bottom

Method 2:
1. Make your window really short
2. Achieve step 2 from Method 1

### Primary pain point(s)

No buttons at the bottom means I can't report bugs 😂 

### Prerequisites

- [X] I have checked to make sure that a similar request has not already been filed or fixed.

rafaellehmkuhl, 2022-04-12:
@ES-Alexander I think I fixed that on #822. Please let me know if this was the case. The image is available under `rafaellehmkuhl/blueos-core:fix_main_menu`.

ES-Alexander, 2022-04-13:
Confirmed - fixes the problem :-)

closed 2022-04-13 by patrickelectric