bluerobotics/BlueOS #792: REC video in QGC problem

fjosen, 2022-02-12:
### Bug description

Hello. 
I can stream video to QGC, but i can`t Rec video. I press the red button and it change from a circle to a rectangel. The "clock" don`t start.

Is that problem in docker or in QGC? do i need to instal more software?

Use:
- Raspberry pi 3b+
- Ardusub
- Pixhawk
- QGC

### Steps to reproduce

.

### Primary pain point(s)

No Video REC is important.

### Prerequisites

- [X] I have checked to make sure that a similar request has not already been filed or fixed.

Williangalvani, 2022-02-14:
Hello!

This seems to be an issue in QGC's side.
It has hard-coded zero-ed timer [here](https://github.com/mavlink/qgroundcontrol/blob/master/src/FlightMap/Widgets/PhotoVideoControl.qml#L287)

Williangalvani, 2022-02-14:
I'm closing it here as it belongs in QGC's repo.
Note that the video is actually recording, it is just the timer that isn't implemented for this case.

closed 2022-02-14 by Williangalvani