bluerobotics/BlueOS #831: bug: Rov lost static ip address setting in latest master

Williangalvani, 2022-03-02:
### Bug description

it used to have 192.168.2.2 as a fixed IP. after the last update I did, it doesn't

### Steps to reproduce

- start in beta12 (13?)
- update to master
- ip should be gone.

### Primary pain point(s)

this means the user has no access to the rov unless it is connected to a wifi network.

### Prerequisites

- [X] I have checked to make sure that a similar request has not already been filed or fixed.

rafaellehmkuhl, 2022-03-02:
Can you upload the cable-guy log? We should know what happened from there.

Williangalvani, 2022-03-04:
I'll try to replicate it soon, I no longer have the logs

patrickelectric, 2022-03-17:
I'm unable to replicate it here

rafaellehmkuhl, 2022-03-17:
> I'm unable to replicate it here

Same

rafaellehmkuhl, 2022-04-01:
Even though we were not able to replicate the issue, if it indeed happened, I believe it was because of a change on the cable-guy base structure (from supporting one IP to supporting many IPs).

Anyway, since we demanded a fresh flash on Beta16/1.0.0, I believe this becomes a non-issue and we can close it.

What do you think, @Williangalvani ?

Williangalvani, 2022-04-01:
Agreed

closed 2022-04-01 by Williangalvani