bluerobotics/BlueOS #839: bug: Service logs are not persistent between docker boots

rafaellehmkuhl, 2022-03-03:
### Bug description

When docker restart, service logs are lost, probably because `/root/.cache/` is not binded.

[image]

### Steps to reproduce

1.  Look at your service logs. There's always just one file. 

### Primary pain point(s)

If one power down BlueOS, the logs are lost.

### Prerequisites

- [X] I have checked to make sure that a similar request has not already been filed or fixed.

rafaellehmkuhl, 2022-03-14:
@Williangalvani for this one, should we add a bind to `startup.json.default`?

Williangalvani, 2022-03-14:
yes

we also need a service/script that udaptes the user current settings.

PR #876 (2022-03-18): core: move service logs to `/var/logs/blueos`
Fix #839 
Fix #861
closed 2022-03-18 by patrickelectric