bluerobotics/BlueOS #840: bug: Successful message on serial firmware upload, but board hangs on bootloader

rafaellehmkuhl, 2022-03-03:
### Bug description

@ES-Alexander reported receiving a success message when trying to upload a firmware to a Pixhawk1.

From what I've check, the only way that could happen was the `uploader.py` script returning a 0 code (all good) on the upload, so maybe that's an issue for the ArduPilot repository, not here.

Anyway, he wasn't able to retry uploading the firmware from BlueOS and did that from QGC, so we should try to support that also from our side.

### Steps to reproduce

We were not able to reproduce it.
We did some firmware upload tries, and they all went well.

### Primary pain point(s)

User gets its board nonoperational.

### Prerequisites

- [X] I have checked to make sure that a similar request has not already been filed or fixed.

ES-Alexander, 2022-05-17:
Seems to have happened again (to someone else) [here](https://discuss.bluerobotics.com/t/water-linked-dvl-a50-support/9123/50).

Agreed that supporting flashing over a non-responsive firmware (like can be done with QGC) would definitely be helpful as a fallback.

rafaellehmkuhl, 2022-05-17:
Copying the workaround here to help while we don't solve this problem:

> Usually one can get around the board-recognition failure by hard-rebooting the companion computer (turning power on and off). The pixhawk board should then appear, so you can try to flash it again with the frontend interface normally. If the flash with the interface does not succeed or the board is not recognized, you can still try to flash the board through the [bultin-terminal](https://docs.bluerobotics.com/ardusub-zola/software/onboard/BlueOS-1.0/advanced-usage/#terminal):
> 
> 1. Open the built-in terminal
> 2. Run ardupilot_fw_uploader.py --identify (to check if the board is being recognized)
> 3. Put your firmware file in the companion computer through the FileBrowser
> 4. Run ardupilot_fw_uploader.py /path/to/firm/file

ES-Alexander, 2022-05-23:
Managed to reproduce the issue today with suitable logging (finally!). 

Found that [the uploader main loop excepts `RuntimeError`s and exits with status 0](https://github.com/ArduPilot/ardupilot/blob/master/Tools/scripts/uploader.py#L1161-L1186). At this point in the code a bootloader is already connected and established, and is going to exit anyway, so should probably have a `sys.exit(1)` in the `except RuntimeError` block (have confirmed that the `finally` block will still be run before the true exit occurs).

Steps from here:
1. Raise PR in ArduPilot repo to fix the issue (so fails are at least detectable)
2. Modify core and frontend to 
   1. alert user that a failed update (once past the detection stage) requires hard rebooting the whole vehicle
   2. persistently store the latest downloaded or uploaded firmware (to support potential install retries after reboots)
   3. provide UI for a "failsafe restore" button that allows attempting to upload the 'persistent' firmware via the uploader without requiring a board to already be detected and connected (will automatically try to connect to a bootloader anyway)
      - may need to change 'installing firmware' UI to have separate downloading and installing stages, so it's obvious that a download has succeeded even if the installation then fails (may also want a separate option to 'download only' or something, but could make for a messier UI for 'normal' usage)
      - failsafe restore can still fail, in which case user needs to reboot system and try again

rafaellehmkuhl, 2022-05-24:
We have to change the commit hash pointing on the `ardupilot_tools/bootstrap.sh` file once the PR gets approved.

rafaellehmkuhl, 2022-05-26:
@ES-Alexander can you check if this was entirely solved (no more sucessful messages on unsucessful uploads)?

ES-Alexander, 2022-05-27:
> @ES-Alexander can you check if this was entirely solved (no more sucessful messages on unsucessful uploads)?

I added a `RuntimeError` to trigger after the first upload progress bar update, and it did report a failure on the frontend but with a seemingly derivative error message rather than the actual one. The RuntimeError's message strangely didn't appear in the log at all, and there were some repeating exceptions afterwards that should probably be cleaned up (perhaps from it trying to connect to an autopilot that wasn't there or something?).

[image]

[logfile_05-26-2022_23_41_20.log](https://github.com/bluerobotics/BlueOS-docker/files/8783058/logfile_05-26-2022_23_41_20.log)

ES-Alexander, 2022-05-27:
Tried again but with some extra messages explicitly sent to `stderr`, and those ones didn't end up in the log. If anything I'd expect `stderr` to be more important than `stdout`, so we should definitely be capturing and logging it.

It may be worth changing [FirmwareUpload.py](https://github.com/bluerobotics/BlueOS-docker/blob/master/core/services/ardupilot_manager/firmware/FirmwareUpload.py#L70-L75) to include something like the following:

```python
            final_error_message = ""
            if process.stderr is not None:
                for line in iter(process.stderr.readline, b""):
                    logger.error(line)
                    final_error_message = line
            while process.poll() is None:
                logger.debug("Waiting for upload tool to finish its job.")
                time.sleep(0.5)
            if process.returncode != 0:
                raise FirmwareUploadFail(f"Upload process returned non-zero code {process.returncode}. {final_error_message}")
            logger.info("Successfully uploaded firmware to board.")
```

rafaellehmkuhl, 2022-05-27:
> I added a `RuntimeError` to trigger after the first upload progress bar update, and it did report a failure on the frontend but with a seemingly derivative error message rather than the actual one. [...]

As you had only one autopilot, and when the flash failed you didn't have it anymore, the only available was SITL, thus this message. The problem is: why the code didn't raise on the FirmwareUploadFail (because of the non-zero return code), as it should?

rafaellehmkuhl, 2022-05-27:
> Tried again but with some extra messages explicitly sent to `stderr`, and those ones didn't end up in the log. If anything I'd expect `stderr` to be more important than `stdout`, so we should definitely be capturing and logging it.
> 
> It may be worth changing [FirmwareUpload.py](https://github.com/bluerobotics/BlueOS-docker/blob/master/core/services/ardupilot_manager/firmware/FirmwareUpload.py#L70-L75) to include something like the following:
> 
> ```python
>             final_error_message = ""
>             if process.stderr is not None:
>                 for line in iter(process.stderr.readline, b""):
>                     logger.error(line)
>                     final_error_message = line
>             while process.poll() is None:
>                 logger.debug("Waiting for upload tool to finish its job.")
>                 time.sleep(0.5)
>             if process.returncode != 0:
>                 raise FirmwareUploadFail(f"Upload process returned non-zero code {process.returncode}. {final_error_message}")
>             logger.info("Successfully uploaded firmware to board.")
> ```
> 
> EDIT: and possibly include a message about probably needing to restart (in the error message), although I'm unsure whether that's Pixhawk only, or all flight controllers.

Good catch. Will add that.

rafaellehmkuhl, 2022-05-27:
> Tried again but with some extra messages explicitly sent to `stderr`, and those ones didn't end up in the log. If anything I'd expect `stderr` to be more important than `stdout`, so we should [...]

I'm not sure if with current implementation we can get both stdout and stderr. Will check.
Maybe this is something to be done with #500

patrickelectric, 2022-08-02:
@rafaellehmkuhl what is the status of this issue ?

rafaellehmkuhl, 2022-08-02:
> @rafaellehmkuhl what is the status of this issue ?

If I recall correctly, this was fixed with #1022. From all my last usages of the firmware uploader, I didn't had the problem again.

patrickelectric, 2022-08-02:
Great, that was I thought.

closed 2022-08-02 by patrickelectric