bluerobotics/BlueOS #958: bug: Reloading the page goes to homepage and not current page

patrickelectric, 2022-04-06:
### Bug description

Reloading the page goes to homepage and not current page 

### Steps to reproduce

Go to any page, hit F5

### Primary pain point(s)

It's annoying to go back to the original page 

### Prerequisites

- [X] I have checked to make sure that a similar request has not already been filed or fixed.

rafaellehmkuhl, 2022-04-06:
This is not happening here.
Were you using companion.local? If so, the `App.vue` redirect condition will redirect you.

patrickelectric, 2022-04-08:
@joaoantoniocardoso I got this from your report, can you explain a bit better ?

joaoantoniocardoso, 2022-04-08:
Sorry, I forgot to mention that this was noticed when I was reviewing the `tools/system-information`, in which if you go to `System Monitor` and hit refresh, it would go back to `Processes`.

rafaellehmkuhl, 2022-04-08:
> Sorry, I forgot to mention that this was noticed when I was reviewing the `tools/system-information`, in which if you go to `System Monitor` (or any other) and hit refresh, it would go back to `Processes`.

Oh, ok. This is because the system-information page is not using routes. I will change the issue title.