bluerobotics/BlueOS #970: Menu text/icons are blurred on chrome.

joaoantoniocardoso, 2022-04-08:
### Current behaviour

On Chrome/Edge, menu fonts and icons are rendering a little blurred, getting sharper when the mouse enters and leaves the menu. It looks always sharp on Firefox.

https://user-images.githubusercontent.com/5920286/162448671-3f6df029-05ea-42f7-a6cd-12f955ca555d.mp4

### Expected or desired behaviour

Icons and texts to be always sharp.

### Prerequisites

- [X] I have checked to make sure that a similar request has not already been filed or fixed.

rafaellehmkuhl, 2022-04-08:
It's fine here:

[image]

Version 100.0.4896.75 (Official Build) (arm64)
MacOS Monterey 12.2.1

Edge is also fine.

rafaellehmkuhl, 2022-04-12:
Apparently this is a bug with the chromium engine. I just updated Edge to `99.0.1150.36` and I'm now seeing the same issue.

patrickelectric, 2024-06-19:
@joaoantoniocardoso let us know if this is a thing

closed 2024-06-19 by patrickelectric

joaoantoniocardoso, 2024-06-20:
Looks all good:
[image]

Firefox (left), Chrome (right)