bluerobotics/BlueOS #991: core: camera-manager: add support for Raspberry Pi cameras

ES-Alexander, 2022-04-20:
### Current behaviour

Raspberry Pi Camera gets plugged into CSI port, but doesn't get detected, and isn't possible to stream to the surface.

Caused by Debian Bullseye switching things away from `v4l2` to `libcamera` (e.g. RPi camera no longer compatible with `v4l2-ctl` or gstreamer's `v4l2src`, and needs to use `libcamerasrc` instead).

Means that users with Raspberry Pi cameras either can't use them, or can't use BlueOS, which is problematic since they're nice as an extra camera option, or a decent camera for low cost vehicle builds.

### Expected or desired behaviour

Ideally should just be able to plug it in and it's detected the same way a H264 USB camera is, and can be streamed to the surface.

### Prerequisites

- [X] I have checked to make sure that a similar request has not already been filed or fixed.

ES-Alexander, 2022-04-21:
On the off chance it could be useful, here's the (unsanitised) terminal output from my unsuccessful attempt at getting this set up and working a month ago (on a standard RPi4 running fresh RaspberryOS Bullseye, not BlueOS):

[unsanitised_attempt_terminal.txt](https://github.com/bluerobotics/BlueOS-docker/files/8526579/unsanitised_attempt_terminal.txt)

patrickelectric, 2022-07-22:
Done.

ES-Alexander, 2024-03-28:
I'm re-opening this because our legacy camera workaround is not guaranteed to work long term, and only supports the original camera module (not the more recent ones, which people are more likely to try to use).

- [Relevant forum thread](https://discuss.bluerobotics.com/t/raspberry-pi-camera-module-3-on-blueos/14984)
- [Relevant camera manager issue](https://github.com/mavlink/mavlink-camera-manager/issues/372)